notro/rpi-source #9: Wrong version headers for 3.18.8-v7+

gcs278, 2015-04-14:
Hey @notro, 

I've been trying to use your script to install the headers for my Raspberry Pi 2 which is using Linux Kernel 3.18.8-v7+ and it seems that the script is downloading linux-headers for 3.18.10+.

When I run your script, it completes successfully. If cd into /linux-7afb1c5b7cf33a3182c97ac9be7379394b9b462a and run <code>make kernelversion</code> I get <code>3.18.10+</code>.

The driver's I am trying to build fail because of the version mismatch.I think it is using the wrong commit hash for the raspberrypi/linux github repo; giving me the wrong version of linux headers.

Thanks

PeterOGB, 2015-04-14:
I will have a look at this tomorrow evening.

PeterOGB, 2015-04-15:
gcs278, Can you run it with --verbose and post the output please ?

notro, 2016-03-30:
Closed, missing feeedback.

closed 2016-03-30 by notro