notro/rpi-source #10: Wrong kernel version string for 3.18.7-v7+

TheCount, 2015-05-04:
Hi @notro,

I tried rpi-source on Raspbian for kernel 3.18.7-v7+, but it builds modules for version 3.18.7+-v7 (note the position of the plus sign). When I try to insmod I get "Invalid module format", presumably because of the version mismatch.

Not sure if this is related to issue #9.

PeterOGB, 2015-05-04:
Please post output when run with debuging enabled.

TheCount, 2015-05-04:
Here is the output:

```
....................................................................................................................................................................................................................
....................................................................................................................................................................................................................
.......................................................................................................................................................  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/auxvec.h
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/hash.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mcs_spinlock.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/scatterlist.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
  GEN     include/generated/bounds.h
  CC      arch/arm/kernel/asm-offsets.s
  GEN     include/generated/asm-offsets.h
  CALL    scripts/checksyscalls.sh
  HOSTCC  scripts/dtc/dtc.o
  HOSTCC  scripts/dtc/flattree.o
  HOSTCC  scripts/dtc/fstree.o
  HOSTCC  scripts/dtc/data.o
  HOSTCC  scripts/dtc/livetree.o
  HOSTCC  scripts/dtc/treesource.o
[... 52 lines cut ...]
sed -i 's/EXTRAVERSION =.*/EXTRAVERSION = +/' /root/linux-0be82f722c097340632a59b879fbfee9c6148f53/Makefile

 *** Create symlink: /root/linux
rm -f /root/linux
ln -s /root/linux-0be82f722c097340632a59b879fbfee9c6148f53 /root/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks
sudo rm -f /lib/modules/$(uname -r)/build /lib/modules/$(uname -r)/source
sudo ln -sf /root/linux /lib/modules/$(uname -r)/build
sudo ln -sf /root/linux /lib/modules/$(uname -r)/source

 *** .config
writef(/root/linux-0be82f722c097340632a59b879fbfee9c6148f53/.config)

 *** Module.symvers
download_to: https://raw.githubusercontent.com/raspberrypi/firmware/8aca5762/extra/Module7.symvers -> /root/linux-0be82f722c097340632a59b879fbfee9c6148f53/Module.symvers
cd /root/linux-0be82f722c097340632a59b879fbfee9c6148f53 && cp -a Module.symvers Module.symvers.backup

 *** make modules_prepare
cd /root/linux && make modules_prepare 

 *** Help: https://github.com/notro/rpi-source/wiki
```

I think the problem is some mixup with EXTRAVERSION and LOCALVERSION.

TheCount, 2015-05-05:
Yes, that's the output of ```rpi-source --verbose```. Stdout anyway. Stderr only contained download progress.

PeterOGB, 2015-05-05:
Ah, sorry, read this on email and didn't scroll down far enough to see the bits I was looking for !PeterO

notro, 2016-03-30:
Fixed https://github.com/notro/rpi-source/commit/f5809e7d744d010b0d6e420dd2659efd238a5851

closed 2016-03-30 by notro