notro/rpi-source #13: No source (Headers) for 4.0.9-v7+

Jasimo, 2015-07-29:
Hi,
I am just trying to use rpi-source for a 4.0.9-v7+ build. Iam getting the follwing after a "sudo rpi-source --skip-gcc":
```
*** rpi-update: https://github.com/Hexxeh/rpi-firmware
 *** Firmware revision: ab7a48a014a52af09df2f7a298bf36d545d3c1fe
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 305, in <module>
    kernel = rpi_update_method(args.uri)
  File "/usr/bin/rpi-source", line 220, in rpi_update_method
    with gzip.open('/proc/config.gz', 'rb') as f:
  File "/usr/lib/python2.7/gzip.py", line 34, in open
    return GzipFile(filename, mode, compresslevel)
  File "/usr/lib/python2.7/gzip.py", line 89, in __init__
    fileobj = self.myfileobj = __builtin__.open(filename, mode or 'rb')
IOError: [Errno 2] No such file or directory: '/proc/config.gz'
```

notro, 2015-07-29:
This results from a change in the kernel config, moving this part into a loadable module.
Run this first: ```sudo modprobe configs```

Maybe @PeterOGB can add this to rpi-source.

Jasimo, 2015-07-29:
works, thx.

plughplover, 2015-08-08:
PLEASE (at a minimum) add that info to wiki, or
add code to try loading module if not present.

notro, 2015-08-08:
wiki updated

commit 5768f95b (2015-08-12): load configs module if necessary

Fixes #13
closed 2015-08-12 by notro