notro/rpi-source #15: can't get all the headers

knezi, 2015-09-05:
Hi, thanks for rpi-source. I am trying to compile module for my TP-link TL-WN715N.
 I tried this:

```shell
pi@raspberrypi ~ $ rpi-source 

 *** gcc version check: OK

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: ea072e1d615bbfeae5ad04f1c9367295dca84f06

 *** Linux source commit: 4507c9752292506fa6ef136114ad14ffd92b2ca5

 *** Download kernel source
--2015-09-05 13:55:42--  https://github.com/raspberrypi/linux/archive/4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz
Resolving github.com (github.com)... 192.30.252.128
Connecting to github.com (github.com)|192.30.252.128|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/4507c9752292506fa6ef136114ad14ffd92b2ca5 [following]
--2015-09-05 13:55:49--  https://codeload.github.com/raspberrypi/linux/tar.gz/4507c9752292506fa6ef136114ad14ffd92b2ca5
Resolving codeload.github.com (codeload.github.com)... 192.30.252.144
Connecting to codeload.github.com (codeload.github.com)|192.30.252.144|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: `/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz'

    [                                                                                                                                                       <=>                ] 128,458,048 1.31M/s   in 4m 29s  

2015-09-05 14:00:25 (466 KB/s) - `/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz' saved [128458048]

 *** Unpack kernel source
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** set 'EXTRAVERSION = +' in Makefile

 *** Create symlink: /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 366, in <module>
    download_to(kernel.symvers, os.path.join(linux_dir, "Module.symvers"))
  File "/usr/bin/rpi-source", line 114, in download_to
    urllib.urlretrieve (url, file)
  File "/usr/lib/python2.7/urllib.py", line 93, in urlretrieve
    return _urlopener.retrieve(url, filename, reporthook, data)
  File "/usr/lib/python2.7/urllib.py", line 239, in retrieve
    fp = self.open(url, data)
  File "/usr/lib/python2.7/urllib.py", line 207, in open
    return getattr(self, name)(url)
  File "/usr/lib/python2.7/urllib.py", line 436, in open_https
    h.endheaders(data)
  File "/usr/lib/python2.7/httplib.py", line 958, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 818, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 780, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 1161, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 553, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
IOError: [Errno socket error] [Errno -2] Name or service not known
```

If I run `rpi-source` again, it says everything OK.
Then trying to compile your hello world example:

```shell
pi@raspberrypi ~/hello $ make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5'

  ERROR: Kernel configuration is invalid.
         include/generated/autoconf.h or include/config/auto.conf are missing.
         Run 'make oldconfig && make prepare' on kernel src to fix it.

  WARNING: Symbol version dump ./Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /home/pi/hello/hello.o
In file included from <command-line>:0:0:
././include/linux/kconfig.h:4:32: fatal error: generated/autoconf.h: No such file or directory
 #include <generated/autoconf.h>
                                ^
compilation terminated.
scripts/Makefile.build:264: recipe for target '/home/pi/hello/hello.o' failed
make[1]: *** [/home/pi/hello/hello.o] Error 1
Makefile:1384: recipe for target '_module_/home/pi/hello' failed
make: *** [_module_/home/pi/hello] Error 2
make: Leaving directory '/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5'
```
Any suggestion what to do?
Thank you.

notro, 2015-09-05:
First delete the folder: /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5
Then run: rpi-source -v
This will let rpi-source unpack the source anew and give some more output.

knezi, 2015-09-07:
Thanks for reply. I do not need this necessarily compile since I get the dongle working, but we can find out what the problem is for other users or potentially for me in the future, If you want.
```shell
pi@raspberrypi ~ $ rpi-source -v
Check for update to rpi-source
download: https://api.github.com/repos/notro/rpi-source/git/refs/heads/master

 *** gcc version check: OK

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: ea072e1d615bbfeae5ad04f1c9367295dca84f06
download: https://raw.githubusercontent.com/Hexxeh/rpi-firmware/ea072e1d615bbfeae5ad04f1c9367295dca84f06/git_hash
sudo modprobe configs 2> /dev/null

 *** Linux source commit: 4507c9752292506fa6ef136114ad14ffd92b2ca5

 *** Download kernel source
wget  -O /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz https://github.com/raspberrypi/linux/archive/4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz
--2015-09-07 08:59:02--  https://github.com/raspberrypi/linux/archive/4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz
Resolving github.com (github.com)... 192.30.252.129
Connecting to github.com (github.com)|192.30.252.129|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/4507c9752292506fa6ef136114ad14ffd92b2ca5 [following]
--2015-09-07 08:59:08--  https://codeload.github.com/raspberrypi/linux/tar.gz/4507c9752292506fa6ef136114ad14ffd92b2ca5
Resolving codeload.github.com (codeload.github.com)... 192.30.252.144
Connecting to codeload.github.com (codeload.github.com)|192.30.252.144|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 128458048 (123M) [application/x-gzip]
Saving to: `/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz'

100%[=========================================================================================================================================================================>] 128,458,048 1.89M/s   in 1m 49s  

2015-09-07 09:01:03 (1.13 MB/s) - `/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz' saved [128458048/128458048]

 *** Unpack kernel source
cd /home/pi && tar --checkpoint=100 --checkpoint-action=dot -xzf /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5.tar.gz
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** set 'EXTRAVERSION = +' in Makefile
sed -i 's/EXTRAVERSION =.*/EXTRAVERSION = +/' /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5/Makefile

 *** Create symlink: /home/pi/linux
rm -f /home/pi/linux
ln -s /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5 /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks
sudo rm -f /lib/modules/$(uname -r)/build /lib/modules/$(uname -r)/source
sudo ln -sf /home/pi/linux /lib/modules/$(uname -r)/build
sudo ln -sf /home/pi/linux /lib/modules/$(uname -r)/source

 *** .config
writef(/home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5/.config)

 *** Module.symvers
download_to: https://raw.githubusercontent.com/Hexxeh/rpi-firmware/ea072e1d615bbfeae5ad04f1c9367295dca84f06/Module.symvers -> /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5/Module.symvers
cd /home/pi/linux-4507c9752292506fa6ef136114ad14ffd92b2ca5 && cp -a Module.symvers Module.symvers.backup

 *** make modules_prepare
cd /home/pi/linux && make modules_prepare 
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
[... 39 lines cut ...]
  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
  CC      arch/arm/kernel/asm-offsets.s
arch/arm/kernel/asm-offsets.c:54:2: error: #error Your compiler is too buggy; it is known to miscompile kernels
 #error Your compiler is too buggy; it is known to miscompile kernels
  ^
arch/arm/kernel/asm-offsets.c:55:2: error: #error and result in filesystem corruption and oopses.
 #error and result in filesystem corruption and oopses.
  ^
Kbuild:61: recipe for target 'arch/arm/kernel/asm-offsets.s' failed
make[1]: *** [arch/arm/kernel/asm-offsets.s] Error 1
Makefile:990: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 371, in <module>
    sh("cd %s && make modules_prepare %s" % (linux_symlink, (" > /dev/null" if args.quiet else "")))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 511, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /home/pi/linux && make modules_prepare ' returned non-zero exit status 2
```

notro, 2015-09-07:
This time rpi-source was able to download Module.symvers.
Now it failed because of the gcc version.

When raspian jessie comes out, I'll see if I can test this and update the wiki with regards to gcc.

knezi, 2015-09-08:
OK, there is gcc 4.8.2 running.

closed 2015-09-08 by knezi

kennyliveforever, 2018-03-25:
```
 *** Unpack kernel source
.......................................tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/powerpc/boot/dts/include: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/powerpc/boot/dts: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/powerpc/boot: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/powerpc: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/mips/boot/dts/include: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/mips/boot/dts: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/mips/boot: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/mips: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/metag/boot/dts/include: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/metag/boot/dts: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/metag/boot: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/metag: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/cris/boot/dts/include: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/cris/boot/dts: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/cris/boot: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/cris: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64/boot/dts/socionext: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64/boot/dts/include: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64/boot/dts/broadcom: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64/boot/dts/arm: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64/boot/dts: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64/boot: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm64: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm/boot/dts/include: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm/boot/dts: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm/boot: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch/arm: Directory renamed before its status could be extracted
tar: linux-a2f34d45809d8685bca1e91989e35746499ac400/arch: Directory renamed before its status could be extracted
tar: Exiting with failure status due to previous errors
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 345, in <module>
    sh("cd %s && tar --checkpoint=100 --checkpoint-action=dot -xzf %s" % (args.dest, linux_tar))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /root && tar --checkpoint=100 --checkpoint-action=dot -xzf /root/linux-a2f34d45809d8685bca1e91989e35746499ac400.tar.gz' returned non-zero exit status 2
```

What it's mean?

sbappudi, 2019-07-23:
getting the same issue as above