notro/rpi-source #16: Wrong kernel version downloaded on Jessie

PeterOGB, 2015-10-04:
I'm just coming back to my project that needs kernel module for an I2S device, and when I run rpi-source on a Jessie machine I end up with sources for 3.18.10 not 4.1.7   I'm investigating.

PeterOGB, 2015-10-04:
First finding... My Jessie machine seems to have the same value in /boot/.firmware_revision as an old Wheezy machine.  That can't be right !

notro, 2015-10-04:
The 'Hello world' example worked fine for me on the 2015-09-24 Jessie image.

Pi2 clean Jessie image without running rpi-update:
```
pi@raspberrypi:~$ cat /boot/issue.txt
Raspberry Pi reference 2015-09-24 (armhf)
Generated using spindle, http://asbradbury.org/projects/spindle/, 4e7ab63, stage4-lxde-edu.qed

pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.1.7-v7+ #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015 armv7l GNU/Linux

pi@raspberrypi:~$ cat /proc/version
Linux version 4.1.7-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #817 SMP PREEMPT Sat Sep 19 15:32:00 BST 2015

pi@raspberrypi:~$ rpi-source
ERROR:
gcc version check: mismatch between gcc (4.9.2) and /proc/version (4.8.3)
Skip this check with --skip-gcc

Help: https://github.com/notro/rpi-source/wiki

pi@raspberrypi:~$ rpi-source --skip-gcc

 *** Using: /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz

 *** Latest firmware revision: 960832a6c2590635216c296b6ee0bebf67b21d50

 *** Linux source commit: 59e76bb7e2936acd74938bb385f0884e34b91d72

 *** Download kernel source
[...]
2015-10-04 20:54:46 (3.03 MB/s) - ‘/home/pi/linux-59e76bb7e2936acd74938bb385f0884e34b91d72.tar.gz’ saved [128488668]

 *** Unpack kernel source
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** set 'EXTRAVERSION = +' in Makefile

 *** Create symlink: /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
[...]
  HOSTCC  scripts/sortextable

 *** ncurses-devel is NOT installed. Needed by 'make menuconfig'. On Raspbian sudo apt-get install libncurses5-dev

 *** Help: https://github.com/notro/rpi-source/wiki

pi@raspberrypi:~$ mkdir hello && cd hello
pi@raspberrypi:~/hello$ nano Makefile
pi@raspberrypi:~/hello$ nano hello.c

pi@raspberrypi:~/hello$ make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/home/pi/linux-59e76bb7e2936acd74938bb385f0884e34b91d72'
  CC [M]  /home/pi/hello/hello.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/pi/hello/hello.mod.o
  LD [M]  /home/pi/hello/hello.ko
make: Leaving directory '/home/pi/linux-59e76bb7e2936acd74938bb385f0884e34b91d72'

pi@raspberrypi:~/hello$ sudo insmod hello.ko
[ 1839.225193] Hello World :)
pi@raspberrypi:~/hello$ sudo rmmod hello
[ 1846.545504] Goodbye World!

```

Pi1 with Jessie image and latest rpi-update kernel:
```
pi@raspberrypi:~$ cat /boot/issue.txt
Raspberry Pi reference 2015-09-24 (armhf)
Generated using spindle, http://asbradbury.org/projects/spindle/, 4e7ab63, stage4-lxde-edu.qed

pi@raspberrypi:~$ uname -a
Linux raspberrypi 4.1.10+ #820 PREEMPT Sun Oct 4 15:33:59 BST 2015 armv6l GNU/Linux

pi@raspberrypi:~$ rpi-source --skip-gcc

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: cad980c560b6c240fdaf6cb4b7703921b18114e3

 *** Linux source commit: 0b439214b09e3f80413c19b7e0a407c34a79411c

 *** Download kernel source
[...]
Saving to: ‘/home/pi/linux-0b439214b09e3f80413c19b7e0a407c34a79411c.tar.gz’

/home/pi/linux-0b439214b09e3f804     [<=>                                                          ] 122.54M  1.29MB/s   in 76s

2015-10-04 20:23:04 (1.60 MB/s) - ‘/home/pi/linux-0b439214b09e3f80413c19b7e0a407c34a79411c.tar.gz’ saved [128490888]

 *** Unpack kernel source
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** set 'EXTRAVERSION = +' in Makefile

 *** Create symlink: /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
[...]
  HOSTCC  scripts/sortextable

 *** ncurses-devel is NOT installed. Needed by 'make menuconfig'. On Raspbian sudo apt-get install libncurses5-dev

 *** Help: https://github.com/notro/rpi-source/wiki

pi@raspberrypi:~$ mkdir hello && cd hello
pi@raspberrypi:~/hello$ nano Makefile
pi@raspberrypi:~/hello$ nano hello.c

pi@raspberrypi:~/hello$ make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/home/pi/linux-0b439214b09e3f80413c19b7e0a407c34a79411c'
  CC [M]  /home/pi/hello/hello.o
  Building modules, stage 2.
  MODPOST 1 modules
  CC      /home/pi/hello/hello.mod.o
  LD [M]  /home/pi/hello/hello.ko
make: Leaving directory '/home/pi/linux-0b439214b09e3f80413c19b7e0a407c34a79411c'

pi@raspberrypi:~/hello$ sudo insmod hello.ko
[  796.656395] Hello World :)

```

PeterOGB, 2015-10-04:
I think the issue may have been a side effect of using rpi-update to get new firmware when the touch screen bug was fixed.  It seemed to have left an out of date .firmware_revision file in /boot  I've just rebuilt the SD card from the .img file and now I have 4.1.7 sources...  Onwards !!!!

closed 2015-10-04 by PeterOGB

criss2525, 2016-03-29:
guys i try to run hello world example on the latest kernel version.
i follow the instructions from here https://github.com/notro/rpi-source/wiki , but i get this. any ideas?
`make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/home/pi/linux-48ba943bfb614ad9a09d4ffd1cfe54adb1550e09'
scripts/Makefile.build:44: /home/pi/hello/Makefile: No such file or directory
make[1]: *** No rule to make target '/home/pi/hello/Makefile'.  Stop.
Makefile:1384: recipe for target '_module_/home/pi/hello' failed
make: *** [_module_/home/pi/hello] Error 2
make: Leaving directory '/home/pi/linux-48ba943bfb614ad9a09d4ffd1cfe54adb1550e09'`