notro/rpi-source #17: Modules end up in wrong place

PeterOGB, 2015-10-05:
I've added my module code into sound/soc/bcm  and sound/soc/codecs and added entries to Kconfig files and Makefiles.  I run

make prepare
make SUBDIRS=sound/soc/bcm  modules
sudo make SUBDIRS=sound/soc/bcm  modules_install
make SUBDIRS=sound/soc/codecs modules
sudo make SUBDIRS=sound/soc/codecs modules_install

but the sound modules end up in 
/lib/modules/4.1.7+-v7/extra

when I think they should be in 
/lib/modules/4.1.7-v7+/kernel/sound/soc/bcm
/lib/modules/4.1.7-v7+/kernel/sound/soc/codecs

Note also the wrong version number "+-v7" vs "-v7+" as well as them all going into "extra" directory.
I'll be investigating this tomorrow evening.
PeterO

notro, 2015-10-05:
The Pi kernel is built from a non vanilla Linux git repository which is the reason a '+' is appended to the version string.
rpi-source on the other hand downloads and unpacks a tarball. This does not trigger the '+' adding.
So rpi-source has to add this '+' for the modules to go in the right directory.

It sets the '+' by editing the kernel Makefile:
```python
info("set 'EXTRAVERSION = +' in Makefile")
sh("sed -i 's/EXTRAVERSION =.*/EXTRAVERSION = +/' %s" % os.path.join(linux_dir, 'Makefile'))
```

However as you have discovered, this obviously doesn't work on Pi2.
It is probably because bcm2709_defconfig sets:
```
CONFIG_LOCALVERSION="-v7"
```
This is not set on Pi1.

So we need to find a way to set that '+' without using EXTRAVERSION.

One way could be to create a ```.scmversion``` file with just a '+' in it in the kernel source root.
[scripts/setlocalversion](https://github.com/raspberrypi/linux/blob/rpi-4.1.y/scripts/setlocalversion) is responsible for assembling the release version string.

PeterOGB, 2015-10-12:
Sorry for not getting back to this sooner... Life gets in the way sometimes :-)   
Using ".scmversion" puts the modules in the right place.  I think them going into the "extra" directory is OK as that is where they are on my Pi1 A+, and modprobe can find them there :-) 

So I think all it needs is to replace the sed with 
with open(".scmversion","w") as f:
    f.write("+")

 PeterO

jacksonliam, 2016-01-20:
Any plans to sort this out?  Got stuck for 3 days trying to figure out what was going wrong! 

I edited the Makefile to remove the + from  EXTRAVERSION and add the + to .scmversion and 'make modules_prepare' and now my module loads properly!

commit f5809e7d (2016-03-30): Use .scmversion to set '+' in kernel release string

Using EXTRAVERSION prevented modules_install from working on Pi2.
The release string got mangled (+-v7 instead of -v7+).
Fix this by putting the + in .scmversion instead.
Fixes #17
closed 2016-03-30 by notro