notro/rpi-source #18: is there a way of updating script for another kernel version?

iamgreenbag, 2016-03-11:
This script works great.. but it's downloading 4.1.13+, when my system's running 4.1.18-v7+. It winds up making an incompatible module. It would be great to point it to a source hosted in Dropbox.

```
greenbag@lotusland:~$ sudo rpi-source --skip-gcc --verbose
Check for update to rpi-source
download: https://api.github.com/repos/notro/rpi-source/git/refs/heads/master

 *** Using: /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz
Latest changes: raspberrypi-firmware (1.20151118-1~wily1.0) wily; urgency=medium

 *** Latest firmware revision: 2a329e0c7d8ea19c085bac5633aa4fccee0f21be
Latest change made by  Serge Schneider <serge@raspberrypi.org>  Wed, 18 Nov 2015 15:59:24 +0000
download: https://raw.githubusercontent.com/raspberrypi/firmware/2a329e0c7d8ea19c085bac5633aa4fccee0f21be/extra/git_hash

 *** Linux source commit: bc1669c846b629cface0aaa367afb2b9c6226faf

 *** Download kernel source
wget  -O /home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf.tar.gz https://github.com/raspberrypi/linux/archive/bc1669c846b629cface0aaa367afb2b9c6226faf.tar.gz
--2016-03-10 20:36:25--  https://github.com/raspberrypi/linux/archive/bc1669c846b629cface0aaa367afb2b9c6226faf.tar.gz
Resolving github.com (github.com)... 192.30.252.131
Connecting to github.com (github.com)|192.30.252.131|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/bc1669c846b629cface0aaa367afb2b9c6226faf [following]
--2016-03-10 20:36:26--  https://codeload.github.com/raspberrypi/linux/tar.gz/bc1669c846b629cface0aaa367afb2b9c6226faf
Resolving codeload.github.com (codeload.github.com)... 192.30.252.161
Connecting to codeload.github.com (codeload.github.com)|192.30.252.161|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 128512095 (123M) [application/x-gzip]
Saving to: ‘/home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf.tar.gz’

/home/greenbag/linux-bc1669c846b629cface0aaa36 100%[===================================================================================================>] 122.56M  1.64MB/s   in 1m 50s 

2016-03-10 20:38:16 (1.12 MB/s) - ‘/home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf.tar.gz’ saved [128512095/128512095]

 *** Unpack kernel source
cd /home/greenbag && tar --checkpoint=100 --checkpoint-action=dot -xzf /home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf.tar.gz
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** set 'EXTRAVERSION = +' in Makefile
sed -i 's/EXTRAVERSION =.*/EXTRAVERSION = +/' /home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf/Makefile

 *** Create symlink: /home/greenbag/linux
rm -f /home/greenbag/linux
ln -s /home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf /home/greenbag/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks
sudo rm -f /lib/modules/$(uname -r)/build /lib/modules/$(uname -r)/source
sudo ln -sf /home/greenbag/linux /lib/modules/$(uname -r)/build
sudo ln -sf /home/greenbag/linux /lib/modules/$(uname -r)/source

 *** .config
writef(/home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf/.config)

 *** Module.symvers
download_to: https://raw.githubusercontent.com/raspberrypi/firmware/2a329e0c7d8ea19c085bac5633aa4fccee0f21be/extra/Module7.symvers -> /home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf/Module.symvers
cd /home/greenbag/linux-bc1669c846b629cface0aaa367afb2b9c6226faf && cp -a Module.symvers Module.symvers.backup

 *** make modules_prepare
cd /home/greenbag/linux && make modules_prepare 
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
#
# configuration written to .config
#
  CHK     include/config/kernel.release
[... 113 lines cut ...]
greenbag@lotusland:~/mt7610u_wifi_sta_v3002_dpo_20130916$ sudo make install
make -C /home/greenbag/mt7610u_wifi_sta_v3002_dpo_20130916/os/linux -f Makefile.6 install
make[1]: Entering directory '/home/greenbag/mt7610u_wifi_sta_v3002_dpo_20130916/os/linux'
mkdir: cannot create directory ‘/etc/Wireless’: File exists
rm -rf /etc/Wireless/RT2860STA
mkdir /etc/Wireless/RT2860STA
cp /home/greenbag/mt7610u_wifi_sta_v3002_dpo_20130916/conf/RT2860STA.dat /etc/Wireless/RT2860STA/.
install -d /lib/modules/4.1.18-v7+/kernel/drivers/net/wireless/
install -m 644 -c mt7650u_sta.ko /lib/modules/4.1.18-v7+/kernel/drivers/net/wireless/
/sbin/depmod -a 4.1.18-v7+
make[1]: Leaving directory '/home/greenbag/mt7610u_wifi_sta_v3002_dpo_20130916/os/linux'
greenbag@lotusland:~$

greenbag@lotusland:~/mt7610u_wifi_sta_v3002_dpo_20130916$ sudo modprobe mt7650u_sta.ko
modprobe: FATAL: Module mt7650u_sta.ko not found.
greenbag@lotusland:~/mt7610u_wifi_sta_v3002_dpo_20130916$ sudo insmod /home/greenbag/mt7610u_*/os/linux/mt7650u_sta.ko
insmod: ERROR: could not insert module /home/greenbag/mt7610u_wifi_sta_v3002_dpo_20130916/os/linux/mt7650u_sta.ko: Invalid module format
greenbag@lotusland:~$

greenbag@lotusland:~$ uname -r
4.1.18-v7+
greenbag@lotusland:~$

linux-bc1669c846b629cface0aaa367afb2b9c6226faf = 4.1.13+
```

notro, 2016-03-12:
rpi-update is meant to be a companion to rpi-update and is made to work with those kernels. However it should work with default Raspian kernels as well.
Yours doesn't seem to match the Raspian I got:
```
~$ zcat /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz | head
raspberrypi-firmware (1.20151118-1) jessie; urgency=medium

  * firmware as of 2a329e0c7d8ea19c085bac5633aa4fccee0f21be

 -- Serge Schneider <serge@raspberrypi.org>  Wed, 18 Nov 2015 15:59:24 +0000
```
I don't think rpi-source will be extended to support more use cases. When there's a Raspian kernel header package in place, rpi-source will most likely be discontinued.

jimlynnjulian, 2017-01-19:
I recently did an upgrade to a Raspbian installation several months old. A driver I compiled, by having to run Rpi-Source first, no longer works. Is there any indication that Rpi-Source usefulness is at an end?

notro, 2017-01-20:
> Is there any indication that Rpi-Source usefulness is at an end?

None that I'm aware of.

jimlynnjulian, 2017-01-20:
Found the problem. Thank you for your response.Jim Julian

closed 2020-08-29 by notro