notro/rpi-source #20: hello world probelm

criss2525, 2016-03-29:
guys i try to run hello world example on the latest kernel version.
i follow the instructions from here https://github.com/notro/rpi-source/wiki , but i get this. any ideas?

pi@raspberrypi:~/hello $ make -C /lib/modules/$(uname -r)/build M=$(pwd) modules
make: Entering directory '/home/pi/linux-48ba943bfb614ad9a09d4ffd1cfe54adb1550e09'
scripts/Makefile.build:44: /home/pi/hello/Makefile: No such file or directory
make[1]: *** No rule to make target '/home/pi/hello/Makefile'.  Stop.
Makefile:1384: recipe for target '_module_/home/pi/hello' failed
make: *** [_module_/home/pi/hello] Error 2

lifanxi, 2016-03-30:
Do you mean this instruction? https://github.com/notro/rpi-source/wiki/Examples-on-how-to-build-various-modules#hello-world-example 

Have you correctly created the Makefile file and hello.c file in you ~/hello directory?

criss2525, 2016-03-30:
yes I mean that instructions. I made the following steps:
1. mkdir hello && cd hello
2. nano makefile 
3. I copied "obj-m := hello.o" in the makefile
4. nano hello.c
5. I copied the C program there
6. make

Did I do something wrong?

criss2525, 2016-03-30:
problem solved: the problem was the first capital letter of the Makefile

closed 2016-03-30 by criss2525