notro/rpi-source #21: Cannot get it running on my Raspberry Pi2 with Jessie (2016-03-18-raspbian-jessie.img)

AlbanT, 2016-04-13:
I followed the installation instructions and it seems to install but when I try to run `sudo rpi-source` I get the message that there is a `mismatch between gcc (4.9.2) and /proc/version (4.7.1)`
```  
pi@raspberrypi:~ $ sudo wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update
    --2016-04-11 20:43:28--  https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source
    Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 23.235.43.133
    Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|23.235.43.133|:443... connected.
    HTTP request sent, awaiting response... 200 OK
    Length: 12471 (12K) [text/plain]
    Saving to: ‘/usr/bin/rpi-source’

    /usr/bin/rpi-source                                         100%[==========================================================================================================================================>]  12.18K  --.-KB/s   in 0s

    2016-04-11 20:43:29 (27.7 MB/s) - ‘/usr/bin/rpi-source’ saved [12471/12471]

    pi@raspberrypi:~ $ sudo rpi-source
    ERROR:
    gcc version check: mismatch between gcc (4.9.2) and /proc/version (4.7.1)
    Skip this check with --skip-gcc

    Help: https://github.com/notro/rpi-source/wiki
    pi@raspberrypi:~ $
```
I can see that this a rather common message as it is featured on the page.
When I run the syntax that is in the *gcc version check* paragraph I get:

```
    pi@raspberrypi:~ $ cat /proc/version
    Linux version 3.18.16-v7 (ubuntu@ubuntu) (gcc version 4.7.1 20120402 (prerelease) (crosstool-NG 1.15.2) ) #4 SMP PREEMPT Thu Sep 10 22:40:21 CST 2015
    pi@raspberrypi:~ $ gcc --version | grep gcc
    gcc (Raspbian 4.9.2-10) 4.9.2
    pi@raspberrypi:~ $
```

But I cannot figure out how to continue :(
Hopefully someone can steer me into the right direction :)

notro, 2016-04-13:
> Skip this check with --skip-gcc

Try: ```sudo rpi-source --skip-gcc```

Maybe this check should have been removed, but there was a time when building a kernel module with the default raspian gcc version gave modules that wouldn't work.

AlbanT, 2016-04-13:
That seemed to work.
But it launched a motherload of questions at me, hope I did well by giving the default answer to each of them?

notro, 2016-04-13:
rpi-source is built to work with offical rpi kernel releases (and beta: rpi-update).
It seems you are using some other kernel?

>  Linux version 3.18.16-v7 (ubuntu@ubuntu) (gcc version 4.7.1 20120402 (prerelease) (crosstool-NG 1.15.2) ) #4 SMP PREEMPT Thu Sep 10 22:40:21 CST 2015

AlbanT, 2016-04-14:
If I interpret the result from 'gcc --version | grep gcc' correctly it is: gcc (Raspbian 4.9.2-10) 4.9.2
I used the most recent jessie image from raspberrypi.org
I didn't have time yet to check if it works.

AlbanT, 2016-04-18:
I got it working using an older image from Raspberry. I used the one from ```2015-11-21-raspbian-jessie``` now instead of the most recent one ```2016-03-18-raspbian-jessie```

closed 2016-04-18 by AlbanT