notro/rpi-source #23: issues building on 4.4

orbitinstasis, 2016-04-22:
I can build on raspbian 4.1, but by bumping to 4.4 (rpi-update) i keep getting the following, even on a fresh copy of raspbian.

pi@raspberrypi:~ $ rpi-source

 *** gcc version check: OK

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: f325bb8328d0839617d66bea8bf396346f76c7a7

 *** Linux source commit: cd560b8658868d9652ab31754e02f86daf6831ba

 *** Download kernel source
--2016-04-22 13:37:39--  https://github.com/raspberrypi/linux/archive/cd560b8658868d9652ab31754e02f86daf6831ba.tar.gz
Resolving github.com (github.com)... 192.30.252.131
Connecting to github.com (github.com)|192.30.252.131|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/cd560b8658868d9652ab31754e02f86daf6831ba [following]
--2016-04-22 13:37:40--  https://codeload.github.com/raspberrypi/linux/tar.gz/cd560b8658868d9652ab31754e02f86daf6831ba
Resolving codeload.github.com (codeload.github.com)... 192.30.252.163
Connecting to codeload.github.com (codeload.github.com)|192.30.252.163|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 136108525 (130M) [application/x-gzip]
Saving to: â/home/pi/linux-cd560b8658868d9652ab31754e02f86daf6831ba.tar.gzâ

/home/pi/linux-cd560b8658868d 100%[================================================>] 129.80M  1.12MB/s   in 2m 7s

2016-04-22 13:39:48 (1.02 MB/s) - â/home/pi/linux-cd560b8658868d9652ab31754e02f86daf6831ba.tar.gzâ saved [136108525/136108525]

 *** Unpack kernel source
..................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** Add '+' to kernel release string

 *** Create symlink: /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf  --silentoldconfig Kconfig
  CHK     include/config/kernel.release
  UPD     include/config/kernel.release
  WRAP    arch/arm/include/generated/asm/bitsperlong.h
  WRAP    arch/arm/include/generated/asm/cputime.h
  WRAP    arch/arm/include/generated/asm/current.h
  WRAP    arch/arm/include/generated/asm/emergency-restart.h
  WRAP    arch/arm/include/generated/asm/errno.h
  WRAP    arch/arm/include/generated/asm/exec.h
  WRAP    arch/arm/include/generated/asm/ioctl.h
  WRAP    arch/arm/include/generated/asm/ipcbuf.h
  WRAP    arch/arm/include/generated/asm/irq_regs.h
  WRAP    arch/arm/include/generated/asm/kdebug.h
  WRAP    arch/arm/include/generated/asm/local.h
  WRAP    arch/arm/include/generated/asm/local64.h
  WRAP    arch/arm/include/generated/asm/mm-arch-hooks.h
  WRAP    arch/arm/include/generated/asm/msgbuf.h
  WRAP    arch/arm/include/generated/asm/msi.h
  WRAP    arch/arm/include/generated/asm/param.h
  WRAP    arch/arm/include/generated/asm/parport.h
  WRAP    arch/arm/include/generated/asm/poll.h
  WRAP    arch/arm/include/generated/asm/preempt.h
  WRAP    arch/arm/include/generated/asm/resource.h
  WRAP    arch/arm/include/generated/asm/rwsem.h
  WRAP    arch/arm/include/generated/asm/seccomp.h
  WRAP    arch/arm/include/generated/asm/sections.h
  WRAP    arch/arm/include/generated/asm/segment.h
  WRAP    arch/arm/include/generated/asm/sembuf.h
  WRAP    arch/arm/include/generated/asm/serial.h
  WRAP    arch/arm/include/generated/asm/shmbuf.h
  WRAP    arch/arm/include/generated/asm/siginfo.h
  WRAP    arch/arm/include/generated/asm/simd.h
  WRAP    arch/arm/include/generated/asm/sizes.h
  WRAP    arch/arm/include/generated/asm/socket.h
  WRAP    arch/arm/include/generated/asm/sockios.h
  WRAP    arch/arm/include/generated/asm/termbits.h
  WRAP    arch/arm/include/generated/asm/termios.h
  WRAP    arch/arm/include/generated/asm/timex.h
  WRAP    arch/arm/include/generated/asm/trace_clock.h
  WRAP    arch/arm/include/generated/asm/unaligned.h
  CHK     include/generated/uapi/linux/version.h
  UPD     include/generated/uapi/linux/version.h
  CHK     include/generated/utsrelease.h
  UPD     include/generated/utsrelease.h
  HOSTCC  scripts/basic/bin2c
  Generating include/generated/mach-types.h
  CC      kernel/bounds.s
  CHK     include/generated/bounds.h
  UPD     include/generated/bounds.h
  CHK     include/generated/timeconst.h
/bin/sh: 1: bc: not found
Kbuild:66: recipe for target 'include/generated/timeconst.h' failed
make[1]: *** [include/generated/timeconst.h] Error 127
Makefile:986: recipe for target 'prepare0' failed
make: *** [prepare0] Error 2
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 373, in <module>
    sh("cd %s && make modules_prepare %s" % (linux_symlink, (" > /dev/null" if args.quiet else "")))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 540, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /home/pi/linux && make modules_prepare ' returned non-zero exit status 2

notro, 2016-04-23:
> /bin/sh: 1: bc: not found

Try ```sudo apt-get install bc```
Then delete the linux-xxxxxxx folder and rerun rpi-source

orbitinstasis, 2016-04-23:
Bingo! Thank you very much for responding, I really appreciate it.

closed 2016-04-26 by orbitinstasis