notro/rpi-source #24: RLError: (SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581)'),)

sh4d9w, 2016-04-26:
Getting always the same issue... what am i doing wrong?

```
root@retropie:~# wget https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source -O /usr/bin/rpi-source && sudo chmod +x /usr/bin/rpi-source && /usr/bin/rpi-source -q --tag-update
--2016-04-26 05:14:58--  https://raw.githubusercontent.com/notro/rpi-source/master/rpi-source
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 185.31.17.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|185.31.17.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 12471 (12K) [text/plain]
Saving to: ‘/usr/bin/rpi-source’

/usr/bin/rpi-source           100%[=================================================>]  12.18K  --.-KB/s   in 0.001s

2016-04-26 05:14:59 (19.8 MB/s) - ‘/usr/bin/rpi-source’ saved [12471/12471]

ERROR:
Couldn't download https://api.github.com/repos/notro/rpi-source/git/refs/heads/master, URLError: (SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581)'),)

Help: https://github.com/notro/rpi-source/wiki
root@retropie:~#
```

notro, 2016-04-27:
I haven't seen that problem with rpi-source.
This is the offending line: https://github.com/notro/rpi-source/blob/master/rpi-source#L104
Maybe google has a solution: https://www.google.no/search?q=urllib2.urlopen+CERTIFICATE_VERIFY_FAILED

sh4d9w, 2016-04-28:
i found a solution here

https://discourse.osmc.tv/t/problems-with-certificates-after-update-to-osmc-2016-01-1/13007/8

seems like u have to make your own certificate....

made-bywok, 2017-02-09:
Just my two cents... check system time. If your raspberry is not time synced, certs are not ( yet ) valid

walkermanX, 2017-11-22:
add agrs {verify=False } to avoid  verification
requests.post(url,params=values,verify=False)

dernasherbrezon, 2020-08-10:
This one might help as well before the script:

```
export PYTHONHTTPSVERIFY=0 
```

closed 2020-08-29 by notro