notro/rpi-source #29: advice for modifying driver

tiibro, 2017-06-16:
Following the instruction in the wiki I can compile and load the hello world driver.

Now I am trying to compile a driver for a touch panel IC (FT5206).  A bit of background on my project can be found here: [https://www.raspberrypi.org/forums/viewtopic.php?t=185948](https://www.raspberrypi.org/forums/viewtopic.php?t=185948).

The source code from the driver is here: [https://github.com/focaltech-systems/drivers-input-touchscreen-FTS_driver](https://github.com/focaltech-systems/drivers-input-touchscreen-FTS_driver) (which is where the manufacturer of my screen pointed me).

I know I need to modify the driver source code so that it compiles and functions on my system, but this is all new to me.  Can anyone offer some advice on learning how to write/modify drivers?  

Also, if there's any specific advice for this IC and driver I'd be very happy for feedback.

Thanks in advance!

notro, 2017-06-16:
Have you tried the in-kernel driver?
https://www.kernel.org/doc/Documentation/devicetree/bindings/input/touchscreen/edt-ft5x06.txt
http://elixir.free-electrons.com/linux/latest/source/drivers/input/touchscreen/edt-ft5x06.c

tiibro, 2017-06-18:
Thanks notro,

After reading this thread: [https://github.com/notro/fbtft/issues/446](https://github.com/notro/fbtft/issues/446), I saw that I needed a `rpi-update` to update the kernel for built in edt_ft5x06 driver support.  The driver is there and a `modinfo edt-ft5x06` gives me reasonable output.

I've tried to make a simple dts overlay using this information: [http://elixir.free-electrons.com/linux/latest/source/Documentation/devicetree/bindings/input/touchscreen/edt-ft5x06.txt](http://elixir.free-electrons.com/linux/latest/source/Documentation/devicetree/bindings/input/touchscreen/edt-ft5x06.txt).  I compile that (saved as edt-ft5x06.dtbo) and add this line: `dtoverlay=edt-ft5x06` to /boot/config.txt.  After rebooting, I do a `lsmod` and see the edt_ft5x06 driver is loaded, but is not used by any hardware.  I can also do a `i2cdetect -y 1` and see that there is a device at address 38.  So the hardware seems to be connected and the driver loaded, but they are not talking.  Also, based on the thread I linked to earlier, I can try `evtest`, but only the mouse and keyboard come up as devices (which seems to make sense as nothing is using the edt_ft5x06 driver).

Attached is my dts file (as a txt).  Any advice would be welcome.  Thanks very much!

[edt-ft5x06.txt](https://github.com/notro/rpi-source/files/1083473/edt-ft5x06.txt)

tiibro, 2017-06-19:
Working on the dts...

My updated file is attached (again as a txt [edt-ft5x06.txt](https://github.com/notro/rpi-source/files/1083880/edt-ft5x06.txt)).  I was missing some parameters which [http://elixir.free-electrons.com/linux/latest/source/Documentation/devicetree/bindings/input/touchscreen/edt-ft5x06.txt](http://elixir.free-electrons.com/linux/latest/source/Documentation/devicetree/bindings/input/touchscreen/edt-ft5x06.txt) reports as being required.  Using this new overlay, the touch panel responds to touch!

I am still confused because an `lsmod` shows the edt_ft5x06 driver loaded, but there is still nothing using the driver.  Maybe I don't understand the output of the command.

Also there are calibration issues with the position of the touch and the position on the screen.  As you see in the attached file, I've added the x and y pixel dimensions to the dts file.  However, this doesn't solve the issue.  I'll start reading up on the tslib package and the ts_test and ts_calibrate tools, but would still welcome any suggestions (especially if the error is in my device tree set-up).

Thanks!

notro, 2017-06-19:
Raspian has changed how X handles input so my knowledge about calibration is moot.

If a program has an open file handle on the input device, it should show up in the lsmod use counter.

tiibro, 2017-06-19:
Thanks again, notro

My mistake was pretty simply, in the dts file I used the LCD TFT screen resolution instead of the touch panel resolution.  Changing this to the correct value, the panel works as expected.  

Thanks for all the help (and for the awesome software).  The discussion has evolved from my simple driver compilation query so again I appreciate your patience and help.

closed 2020-08-29 by notro