notro/rpi-source #33: Building module using kernel sources results in message that the module taints the kernel

hgbutte, 2017-10-27:
Does this mean there's a mismatch between the kernel version i use (got thru rpi-update) and the kernel source i have obtained using rpi-source ?

Tried this with 4.9.57+ and 4.9.58+ on a Raspi 1.

That's the module. I needed to modify the source to compile for 4.9.x:
https://www.ixxat.com/de/produkte/industrie-produkte/pc-interfaces/pc-can-interfaces/socketcan

notro, 2017-10-27:
Out-of-tree modules taint the kernel.
https://unix.stackexchange.com/questions/118116/linux-what-is-a-tainted-kernel
http://elixir.free-electrons.com/linux/latest/ident/check_modinfo

hgbutte, 2017-10-29:
Thx for reply.
I have had issues using the driver causing the device to re-initialize from time to time which i haven't seen before. So i thought it might be an issue with the sources.

closed 2020-08-29 by notro