notro/rpi-source #36: gzip: stdin: invalid compressed data--format violated

nick-jonas, 2018-01-05:
I'm trying this on a fresh version of Raspbian from the [official download site](https://www.raspberrypi.org/downloads/raspbian/):

RASPBIAN STRETCH LITE
Minimal image based on Debian Stretch
Version:November 2017
Release date:2017-11-29
Kernel version:4.9

System info:
```
$ cat /proc/version
Linux version 4.9.73+ (dc4@dc4-XPS13-9333) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611) ) #1072 Sun Dec 31 19:28:20 GMT 2017
```

When trying to run ```rpi-source```:

```
$ rpi-source --skip-gcc

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: 9fdd5ca79b8fea585ad92b36190158cca2b07eb6

 *** Linux source commit: fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1

 *** Download kernel source: Already downloaded /home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz

 *** Unpack kernel source
...................................................................................................................................................tar: Skipping to next header
.
gzip: stdin: invalid compressed data--format violated
tar: Child returned status 1
tar: Error is not recoverable: exiting now
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 345, in <module>
    sh("cd %s && tar --checkpoint=100 --checkpoint-action=dot -xzf %s" % (args.dest, linux_tar))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 186, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /home/pi && tar --checkpoint=100 --checkpoint-action=dot -xzf /home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz' returned non-zero exit status 2

notro, 2018-01-05:
I guess your file is corrupt somehow, at least it works for me. Delete the file (or rename) and it will be downloaded again.
```
pi@pi3:~ $ sudo rpi-update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** Performing self-update
 *** Relaunching after update
 *** Raspberry Pi firmware updater by Hexxeh, enhanced by AndrewS and Dom
 *** We're running for the first time
 *** Backing up files (this will take a few minutes)
 *** Remove old firmware backup
 *** Backing up firmware
 *** Remove old modules backup
 *** Backing up modules 4.15.0-rc2-00006-gd88f6e371563-dirty
#############################################################
This update bumps to rpi-4.9.y linux tree
Be aware there could be compatibility issues with some drivers
Discussion here:
https://www.raspberrypi.org/forums/viewtopic.php?f=29&t=167934
##############################################################
 *** Downloading specific firmware revision (this will take a few minutes)
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100   168    0   168    0     0    248      0 --:--:-- --:--:-- --:--:--   248
100 54.1M  100 54.1M    0     0  1084k      0  0:00:51  0:00:51 --:--:-- 1148k
 *** Updating firmware
 *** Updating kernel modules
 *** depmod 4.9.73+
 *** depmod 4.9.73-v7+
 *** Updating VideoCore libraries
 *** Using HardFP libraries
 *** Updating SDK
 *** Running ldconfig
 *** Storing current firmware revision
 *** Deleting downloaded files
 *** Syncing changes to disk
 *** If no errors appeared, your firmware was successfully updated to 9fdd5ca79b8fea585ad92b36190158cca2b07eb6
 *** A reboot is needed to activate the new firmware
pi@pi3:~ $ sudo reboot

pi@pi3:~ $ rpi-source --skip-gcc

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: 9fdd5ca79b8fea585ad92b36190158cca2b07eb6

 *** Linux source commit: fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1

 *** Download kernel source
--2018-01-05 21:49:10--  https://github.com/raspberrypi/linux/archive/fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1 [following]
--2018-01-05 21:49:11--  https://codeload.github.com/raspberrypi/linux/tar.gz/fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1
Resolving codeload.github.com (codeload.github.com)... 192.30.253.120, 192.30.253.121
Connecting to codeload.github.com (codeload.github.com)|192.30.253.120|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz’

/home/pi/linux-fbcbd7e7494c6b04cf9e2     [                             <=>                                     ] 138.43M  4.24MB/s    in 57s

2018-01-05 21:53:46 (2.41 MB/s) - ‘/home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz’ saved [145155263]

 *** Unpack kernel source
.................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** Add '+' to kernel release string

 *** Create symlink: /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
<snip>
  HOSTCC  scripts/sortextable

 *** ncurses-devel is NOT installed. Needed by 'make menuconfig'. On Raspbian sudo apt-get install libncurses5-dev

 *** Help: https://github.com/notro/rpi-source/wiki
pi@pi3:~ $

```

nick-jonas, 2018-01-05:
Could the fact that I'm using a Pi Zero make a difference?

  I've tried deleting the file and redoing it several times, but getting the same type of error (this one shows "Unexpected EOF in archive"):

```
pi@raspberrypizero:~ $ rpi-source --skip-gcc

 *** rpi-update: https://github.com/Hexxeh/rpi-firmware

 *** Firmware revision: 9fdd5ca79b8fea585ad92b36190158cca2b07eb6

 *** Linux source commit: fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1

 *** Download kernel source
--2018-01-05 23:19:16--  https://github.com/raspberrypi/linux/archive/fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1 [following]
--2018-01-05 23:19:16--  https://codeload.github.com/raspberrypi/linux/tar.gz/fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1
Resolving codeload.github.com (codeload.github.com)... 192.30.253.121, 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 145155263 (138M) [application/x-gzip]
Saving to: ‘/home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz’

/home/pi/linux-fbcbd7e7494c6b04cf9e2b 100%[========================================================================>] 138.43M  2.04MB/s    in 79s

2018-01-05 23:20:36 (1.76 MB/s) - ‘/home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz’ saved [145155263/145155263]

 *** Unpack kernel source
.....................................................................................................................................................................................................
gzip: stdin: invalid compressed data--format violated
tar: Unexpected EOF in archive
tar: Unexpected EOF in archive
tar: Error is not recoverable: exiting now
Traceback (most recent call last):
  File "/usr/bin/rpi-source", line 345, in <module>
    sh("cd %s && tar --checkpoint=100 --checkpoint-action=dot -xzf %s" % (args.dest, linux_tar))
  File "/usr/bin/rpi-source", line 85, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 186, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /home/pi && tar --checkpoint=100 --checkpoint-action=dot -xzf /home/pi/linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz' returned non-zero exit status 2
```

notro, 2018-01-05:
Maybe the sd card is faulty?
This is the checksum:
```
pi@pi3:~ $ md5sum linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz
9c3135ec6271df3ebb5a5909244b63ff  linux-fbcbd7e7494c6b04cf9e2b2c955ce7d3a9335bb1.tar.gz
```

nick-jonas, 2018-01-08:
That was totally it - redid everything with a new SD card.  Thanks!!

closed 2018-01-08 by nick-jonas