notro/rpi-source #37: rpi-source adds a "+" to the kernel.release string

careyer, 2018-01-17:
Hi there,

I want to add a kernel module in rabian jessie (4.9.35-v7)... when I download the sources via rpi-source it says during the process "*** Add "+" to kernel release string" 

When I now make and make install my module it will compile for kernel-version 4.9.35-v7+ and not run on my actual kernel 4.9.35-v7

How can I solve this problem?
Thank you so much for helping me out!

Cheers!

notro, 2018-01-18:
rpi-source is made to work with rpi-update installed kernels which have a + in the kernel name:
https://github.com/raspberrypi/firmware/blob/master/extra/uname_string
This is the copy installed by rpi-update:
https://github.com/Hexxeh/rpi-firmware/tree/master/modules

The + is applied when building a kernel from a git repo source.

You can avoid this my commentingout/removing a couple of lines in /usr/bin/rpi-source:
https://github.com/notro/rpi-source/blob/master/rpi-source#L350

careyer, 2018-01-18:
@notro Thank you for comming back to my request in no time and for providing invaluable information. I managed to get rid of the "+" manually. I did a grep command and searched the whole filesystem for any occurrence of the altered kernel.version string. Now my kernel module compiles with the correct kernel version an will run without problems!

Thank you very much!
Take Care!

closed 2018-01-18 by careyer