notro/rpi-source #40: rpi-source seems to pull the wrong kernel version on Raspbian 2018-03-13?

rodizio1, 2018-11-21:
Don't really have an explanation for this. I'm trying to pull the kernel sources using rpi-source on Raspbian 2018-03-13 which uses Kernel 4.9.80. However, rpi-source somehow pulls Kernel version 4.14.24. I have used rpi-source with other Raspbian versions in the past, it always pulled the correct corresponding kernel version so far.

Here is what's happening (fresh Raspbian 2018-03-13 image with no changes):

Running kernel is 4.9.80
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
```

rpi-source pulling commit 80a14a56dacb7cc2b40d5f37d00bedb0ceace130
```
pi@raspberrypi:~ $ rpi-source --skip-gcc

 *** Using: /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz

 *** Latest firmware revision: 3347884c7df574bbabeff6dca63caf686e629699

 *** Linux source commit: 80a14a56dacb7cc2b40d5f37d00bedb0ceace130

 *** Download kernel source
--2018-11-21 09:29:17--  https://github.com/raspberrypi/linux/archive/80a14a56dacb7cc2b40d5f37d00bedb0ceace130.tar.gz
Resolving github.com (github.com)... 192.30.253.112, 192.30.253.113
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/80a14a56dacb7cc2b40d5f37d00bedb0ceace130 [following]
--2018-11-21 09:29:18--  https://codeload.github.com/raspberrypi/linux/tar.gz/80a14a56dacb7cc2b40d5f37d00bedb0ceace130
Resolving codeload.github.com (codeload.github.com)... 192.30.253.121, 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/home/pi/linux-80a14a56dacb7cc2b40d5f37d00bedb0ceace130.tar.gz’
```

Kernel sources downloaded and extracted:
```
pi@raspberrypi:~ $ ls -l
total 155928
drwxr-xr-x 24 pi pi      4096 Nov 21 09:33 linux-80a14a56dacb7cc2b40d5f37d00bedb0ceace130
-rw-r--r--  1 pi pi 159658880 Nov 20 21:24 linux-80a14a56dacb7cc2b40d5f37d00bedb0ceace130.tar.gz
```

Looking at the version shows 4.14.24:
```
pi@raspberrypi:~ $ cd linux-80a14a56dacb7cc2b40d5f37d00bedb0ceace130/
pi@raspberrypi:~/linux-80a14a56dacb7cc2b40d5f37d00bedb0ceace130 $ make kernelversion
4.14.24
```

But we're running Raspbian 2018-03-13 with kernel 4.9.80:
```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.9.80-v7+ #1098 SMP Fri Mar 9 19:11:42 GMT 2018 armv7l GNU/Linux
```

rodizio1, 2018-11-21:
Well, I just looked into /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz and it says "Breaks utilities which use firmware commit to determine kernel commit" right at the top. This probably explains the issue (?) Is there another way to find out the correct commit id to download 4.9.80 sources?

> raspberrypi-firmware (1.20180313-1) stretch; urgency=medium
> 
>   * firmware as of 3347884c7df574bbabeff6dca63caf686e629699
>   * Kernel 4.9.80
>     - Breaks utilities which use firmware commit to determine kernel commit
> 
>  -- Serge Schneider <serge@raspberrypi.org>  Tue, 13 Mar 2018 21:05:21 +0000

notro, 2018-11-23:
What's the top of that changelog file? That's where rpi-source looks.

I have just tried the latest raspbian and it works for me:
```
pi@pi2835:~$ cat /boot/issue.txt
Raspberry Pi reference 2018-11-13
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 7e0c786c641ba15990b5662f092c106beed40c9f, stage5
```

```
pi@pi2835:~$ zcat /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz | head
raspberrypi-firmware (1.20181112-1) stretch; urgency=medium

  * firmware as of 12e0bf86e08d6067372bc0a45d7e8a10d3113210

 -- Serge Schneider <serge@raspberrypi.org>  Mon, 12 Nov 2018 17:26:39 +0000

raspberrypi-firmware (1.20180924-1) stretch; urgency=medium

  * firmware as of 5b49caa17e91d0e64024380119ad739bb201c674

```

```
pi@pi2835:~$ python work/rpi-source/rpi-source --skip-update

 *** Using: /usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz

 *** Latest firmware revision: 12e0bf86e08d6067372bc0a45d7e8a10d3113210

 *** Linux source commit: 36612d5d7a88672a3e7dd6cb458dbbbca0d75efe

 *** Download kernel source
--2018-11-23 17:17:57--  https://github.com/raspberrypi/linux/archive/36612d5d7a88672a3e7dd6cb458dbbbca0d75efe.tar.gz
Resolving github.com (github.com)... 140.82.118.4, 140.82.118.3
Connecting to github.com (github.com)|140.82.118.4|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://codeload.github.com/raspberrypi/linux/tar.gz/36612d5d7a88672a3e7dd6cb458dbbbca0d75efe [following]
--2018-11-23 17:17:58--  https://codeload.github.com/raspberrypi/linux/tar.gz/36612d5d7a88672a3e7dd6cb458dbbbca0d75efe
Resolving codeload.github.com (codeload.github.com)... 192.30.253.121, 192.30.253.120
Connecting to codeload.github.com (codeload.github.com)|192.30.253.121|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: unspecified [application/x-gzip]
Saving to: ‘/home/pi/linux-36612d5d7a88672a3e7dd6cb458dbbbca0d75efe.tar.gz’

/home/pi/linux-36612d5d7a88672a3     [             <=>                                           ] 152.52M  2.01MB/s    in 92s

2018-11-23 17:19:31 (1.65 MB/s) - ‘/home/pi/linux-36612d5d7a88672a3e7dd6cb458dbbbca0d75efe.tar.gz’ saved [159929806]

 *** Unpack kernel source
...............................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................................
 *** Add '+' to kernel release string

 *** Create symlink: /home/pi/linux

 *** Create /lib/modules/<ver>/{build,source} symlinks

 *** .config

 *** Module.symvers

 *** make modules_prepare
  HOSTCC  scripts/basic/fixdep
<snip>
  HOSTCC  scripts/sortextable

 *** ncurses-devel is NOT installed. Needed by 'make menuconfig'. On Raspbian sudo apt-get install libncurses5-dev

 *** Help: https://github.com/notro/rpi-source/wiki
pi@pi2835:~$
```

rodizio1, 2018-11-26:
The changelog snippet I posted was the start if I remember correctly.

Yes, rpi-source worked fine with any Raspbian version that I tried so far, only on version 2018-03-13 it pulls the wrong kernel sources.

notro, 2018-12-02:
The only reason I can think of is that it's a copy/paste error in the changelog.
The mentioned fw commit does point to 4.14.24: https://github.com/raspberrypi/firmware/blob/3347884c7df574bbabeff6dca63caf686e629699/extra/uname_string

MichaIng, 2019-01-07:
@notro 
Just a quick question:
- Does `rpi-source` always install headers for the lastest Raspbian package and `rpi-firmware` (`rpi-update`) version, or does it try to install headers for the currently installed kernel as well?
- So if one ran `rpi-update` in the past and needs the headers for his installed version, is it required to run `rpi-update` first, to not run into a mismatch?

notro, 2019-01-07:
rpi-source was made at a time where there were no kernel header package for the Pi. There is now (Raspbian), but for some reason people still use rpi-source. Maybe because of all the howto's that use it, or that their distro doesn't have a kernel header package.

In this day and age rpi-source is really a companion to rpi-update.
There is one use case for rpi-source even when there's a header package and that is if someone wants to build an in-kernel module that isn't enabled.

This is what rpi-source does:
- If there is a ```/boot/.firmware_revision```, use this firmware commit.
- Otherwise it checks ```/usr/share/doc/raspberrypi-bootloader/changelog.Debian.gz``` to find the firmware commit. I believe that installing a updated kernel package will update the changelog.
- Having the firmware commit, it looks up the kernel source commit and downloads the archive

I haven't tried, but I believe it should handle the next branch, since it only cares about the commit hash.

MichaIng, 2019-01-07:
@notro 
Perfect, thanks for the info. Jep then my aim is quite simple to achieve:
- If `rpi-update` is installed, use `rpi-source` to assure the fitting headers are installed.
- Otherwise assume that Raspbian kernel package is installed and install the related headers package via APT.

closed 2020-08-29 by notro