notro/rpi-source #43: General source path

hawkeyexp, 2019-05-04:
Hi,

first: a great script !

But one point: why not using /usr/src for example to place the kernel source instead of placing it into /root home dir ?

notro, 2019-05-04:
The idea is to run the script as the pi user, _not_ as root or with sudo. The source is put in the home folder. /user/src is read-only for regular users. You can override destination with ```-d <folder>```. The script uses sudo internally in order to create a couple of symlinks. rpi-source was created to make it easy for novices to build out-of-tree modules. Nowadays there's a kernel header package, but still people finds use for it. Maybe because they use old howtos that hasn't been updated? I don't know.

kapitainsky, 2019-06-27:
kernel header package is often behind the latest kernel releases so this script is mega useful.

closed 2020-08-29 by notro