notro/rpi-source #47: fatal error: openssl/bio.h: Datei oder Verzeichnis nicht gefunden

Trackhe, 2020-02-09:
scripts/extract-cert.c:21:10: fatal error: openssl/bio.h: Datei oder Verzeichnis nicht gefunden
 #include <openssl/bio.h>
          ^~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [scripts/Makefile.host:90: scripts/extract-cert] Fehler 1
make: *** [Makefile:1061: scripts] Fehler 2
Traceback (most recent call last):
  File "./rpi-source", line 411, in <module>
    sh("cd %s && make modules_prepare %s" % (linux_symlink, (" > /dev/null" if args.quiet else "")))
  File "./rpi-source", line 90, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /root/linux && make modules_prepare ' returned non-zero exit status 2

Cysioland, 2020-05-18:
Make sure you installed `libssl-dev`

Trackhe, 2020-05-19:
Yes this is also the first think what i do after getting the error but i dont know what happend. It is 2 moth ago and i have fix it, i dont know how but anyway. I close this.

closed 2020-05-19 by Trackhe