notro/rpi-source #50: gcc: error: unrecognized command line option ‘-mgeneral-regs-only’

Cysioland, 2020-05-18:
When running rpi-source I get:

```
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
make[1]: *** [scripts/Makefile.build:266: scripts/mod/empty.o] Error 1
make: *** [Makefile:1110: prepare0] Error 2
Traceback (most recent call last):
  File "/usr/local/bin/rpi-source", line 412, in <module>
    sh("cd %s && make modules_prepare %s" % (linux_symlink, (" > /dev/null" if args.quiet else "")))
  File "/usr/local/bin/rpi-source", line 91, in sh
    subprocess.check_call(cmd, shell=True)
  File "/usr/lib/python2.7/subprocess.py", line 190, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command 'cd /root/linux && make modules_prepare ' returned non-zero exit status 2
```

My kernel is `5.4.40-v8+ #1316 SMP PREEMPT Tue May 12 13:17:34 BST 2020 aarch64 GNU/Linux`

notro, 2020-06-03:
I'm sorry but I have to much to do at the moment to look into this now. If someone know what the problem might be, please do chime in.

satmandu, 2020-06-06:
I think this is because you're running with a 32 bit userland and a 64 bit kernel.

Try running this with the 64 bit userland, like in https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370

Cysioland, 2020-06-06:
I already swapped to a regular 64-bit Ubuntu compilation for RPi and it works turnkey, basically.

dony71, 2020-06-16:
Is it possible to use  aarch64-linux-gnu-gcc (cross compile) ?
If possible, which option should I put?

satmandu, 2020-06-16:
https://packages.debian.org/buster/gcc-aarch64-linux-gnu doesn't appear to show any packages for aarch64-linux-gnu-gcc on armhf.

You might be better off at this point trying out the new arm64 raspbian distribution which is now in beta. See https://www.raspberrypi.org/forums/viewtopic.php?f=117&t=275370

closed 2020-08-29 by notro